espressif/ESP8266_MP3_DECODER #4: Link to 23LC1024  and ES9023

confile, 2015-11-01:
Could you please post a link to the needed hardware. It is hard to find what is needed.

Spritetm, 2015-11-01:
What exactly do you need? You can obtain both parts from electronic wholesalers like Digikey and Farnell. The 23LC1024 is directly usable; see the readme.md in the archive. The ES9023 needs some extra components; for that, look at the datasheet or get a pre-made board with the chip on it like eg (this)[http://www.ebay.nl/itm/Audiophonics-DAC-Sabre-ES9023-I2S-vers-Analogique-24bit-192KHZ-fur-Raspberry-PI-/291573327639?hash=item43e31fe717:g:TTwAAOSw~OdVZXtm] one. (Note: auction was randomly chosen, we have no experience with this specific seller.)

confile, 2015-11-01:
Could you please post the ebay links to all components?

What I want to do is the following. Add a button to the ESP8266 and a small speaker. When clicking the button the modules connects to the internet and loads a voice mp3 file about 30 sek. and plays it. 

What would you recommend for that?

Spritetm, 2015-11-01:
I cannot; because my location differs from yours, the offering and shipping prices may differ vastly. I advise you to use Ebays search feature. Sorry.

closed 2015-11-01 by Spritetm

confile, 2015-11-01:
Ebay nl would also be fine. I just need the correct name to search for.

confile, 2015-11-01:
@Spritetm Please help me what do I have to search for?