espressif/ESP8266_MP3_DECODER #5: Woking with Nodemcu?

confile, 2015-11-02:
Does this project also wok with Nodemcu?

http://www.nodemcu.com/index_en.html

Spritetm, 2015-11-03:
I highly doubt it. The project uses FreeRTOS while NodeMCU (afaik) uses the os-less SDK. Besides that, the MP3 decoder has almost no spare memory to allow NodeMCU to run in.

closed 2015-11-03 by Spritetm

confile, 2015-11-03:
@Spritetm is it possible to use a NodeMCU firmware and and some external memory to build a mp3 decoder. Please give me some hint on that.

confile, 2015-11-03:
@Spritetm Or is there an external decoder module I can use?

Spritetm, 2015-11-03:
Maybe, if you can convince NodeMCU to put everything it has into external RAM. That's outside the scope of this project, though. And yes, there are a bunch of external MP3 modules available on the market. Choosing one of those also is outside the scope of this project.