espressif/ESP8266_MP3_DECODER #9: What does the 23LC1024 while programming?

milkpirate, 2015-11-24:
Hey,

as suggested I piggyback the 23LC1024 onto the flash chip of my ESP-12E as suggested in the README.md. I also connected the /CE pin of the 23LC1024 to GPIO0.

So to program the ESP (i.e. enter the bootloader) the GPIO0 pin has to be low but since it is also connected to the /CE of the 23LC1024 (which is active low as well) the RAM chip is enabled while programming and might/does interfer with the ESP talking to the flash chip.

I also tired removing the pulldown of the GPIO0 after entering the bootloader but the programming still fails!

Please tell me if I have a twisted way of thinking...

Spritetm, 2015-11-24:
GPIO0 does only need to be low during the time the ESP8266 is reset, and most implementations of auto-reset and push-button-to-program will do exactly that. During programming, GPIO0 can (and seemingly needs to be, as you found out) high and the 23LC1024 will be idle because of that. 

This is a quirk worth documenting, though. I'll put it in the Readme file once I find some time for that.

milkpirate, 2015-11-28:
Thank you, Spritetm for you consideration and time. I will try again with diconnected /CE.

I also have another question regarding the make line:
make COMPILE=gcc BOOT=none APP=0 SPI_SPEED=40 SPI_MODE=QIO SPI_SIZE=1024

I know my EPS-12E has a 4Mbit (=4096 Kbit) SPI flash (datesheet) so what unit does the the SPI_SIZE parameter take? Is it Bit or Byte? So shall I change it to 512 (Kbyte) or 4096 (Kbit) to be consistent with my module? I once already had problems with programming an ESP-01 caused by wrong programming parameters so that might also be the reason for failing in programming.

Spritetm, 2015-11-30:
The size is in KByte, so you would have to use 512 there. DIO or QIO is dependent on your ESP module; I know some versions of the ESP12E need DIO. If you have successfully flashed and ran other code using QIO into the module, that should work for this code too. Whatever you use, make sure your flashing program also uses that mode (most will default to QIO).

milkpirate, 2015-12-07:
Ok. I tried soo many combination of DIO, QIO and sizes and eveytime I get 

    <rubbish>MEM CHECK FAIL!!!

I use the `make.sh` in the `mp3` folder:

```
#!/bin/bash -e

SPI_MODE=DIO
SPI_SIZE=4096

make clean
make COMPILE=gcc BOOT=none APP=0 SPI_SPEED=40 SPI_MODE=$SPI_MODE SPI_SIZE=$SPI_SIZE

ESPTOOL=/home/lenny/mcu/esp-open-sdk/esptool/esptool.py
ESPBAUD=115200
ESPPORT=/dev/ttyUSB0

echo "make COMPILE=gcc BOOT=none APP=0 SPI_SPEED=40 SPI_MODE=$SPI_MODE SPI_SIZE=$SPI_SIZE"

${ESPTOOL} --port ${ESPPORT} --baud ${ESPBAUD} write_flash 0x00000 ../bin/eagle.flash.bin 0xA0000 ../bin/eagle.irom0text.bin
```

As far as I see theres no options in the ESPTOOL commandline to decide what SPI moder it should use. Could my module be broken?

Spritetm, 2015-12-08:
Hi,

Ah, the MEM CHECK FAIL thing is a big red herring... it is generated by 
the ROM in the ESP itself if you use EXT_RESET as a reset pin instead of 
CH_PD. It has nothing to do with the SPI RAM.

Cheers,
Jeroen

Spritetm, 2015-12-08:
Also, esptool does have options to set DIO etc but it's hidden away behind the write_flash subcommand. Run esptool.py write_flash --help for the options.

milkpirate, 2015-12-10:
Ok well that might be not a topic of the RAM anymore but I really would like to get this working. I would really appreciate it if you could go on helping me.

So what I did was to add a 10k pullup resistor between VCC and GPIO0 not to leave it floating and undefined after reset to bootloader mode. I switched from using ESPTOOL to the nodemcu flasher where if have the options conveniently avaiable. So copied the files and flashed then 1st in DIO mode to:

```
0x00000 <= eagle.flash.bin
0xA0000 <= eagle.irom0text.bin
```
which resultet in a bootloop:

```
▒MEM CHECK FAIL!!!
Fatal exception (0):
epc1=0x402bbae1
epc2=0x40100070
epc3=0x00000000
epcvaddr=0x00000000
depc=0x402a0208
rtn_add=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0
tail 12
chksum 0xde
csum 0xde
▒MEM CHECK FAIL!!!
Fatal exception (0):
epc1=0x402bbae1
epc2=0x40100070
epc3=0x00000000
epcvaddr=0x00000000
depc=0x402a0208
rtn_add=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0
tail 12
chksum 0xde
csum 0xde
...
```

2nd try: QIO (same bootloop)
```
▒ՁMEM CHECK FAIL!!!
Fatal exception (0):
epc1=0x402bbae1
epc2=0x40100070
epc3=0x00000000
epcvaddr=0x00000000
depc=0x402a0208
rtn_add=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0
tail 12
chksum 0xde
csum 0xde
▒ՁMEM CHECK FAIL!!!
Fatal exception (0):
epc1=0x402bbae1
epc2=0x40100070
epc3=0x00000000
epcvaddr=0x00000000
depc=0x402a0208
rtn_add=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0
tail 12
chksum 0xde
csum 0xde
...
```

PS: I was wrong about the size of my flash it was 32Mbit = 4Mbyte. So `SPI_SIZE=4096`.

Spritetm, 2015-12-15:
Seems like a power supply issue to me, to be honest. Do you use a good 3,3V power supply, and enough decoupling caps?

milkpirate, 2015-12-15:
Hey, 

well I thought so. I already had 100nF, 10uF caps there but as you suggested I added ridiculous amount more to it and surprise, surprise it works a bit better but still kinda unstable:

(log shortened)
```
MEM CHECK FAIL!!!
ä€òžàŽrnnœnŒìŒœœœrbbnn‚ìbì‚rrn’lŒl`€’nžàŽrnnœnŒìŒœœœŽŽràbbnn‚bì‚rrnŒl`rnòžàŽrnnœnŒìŒœœœŽp€bbnn‚b|rrnŽŽlb‚‚nâþ‚rrnnœnŒìŒœœœŒŒbbnn€bì‚rrn‚ìŽŽlnîþ‚rrnnœnŒìŒœœœlŽpàbbnnŒb|rrn’lŒl`à’n|ržàŒ‚r’p`’‚’‚nâŒ’`brbì÷

Hardware initialized. Waiting for network.
mode : sta(18:fe:34:xx:xx:xx)
add if0
scandone
add 0
aid 6
cnt 

connected with WIFI_STATION, channel 1
dhcp client start...
ip:192.168.0.22,mask:255.255.255.0,gw:192.168.0.1
Fatal exception (0): 
epc1=0x402c3b3f
epc2=0x00000000
epc3=0x402a6307
epcvaddr=0x00000000
depc=0x00000000
rtn_add=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16 
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0 
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0 
tail 12
chksum 0xde
csum 0xde
Œ‚ÿ
MEM CHECK FAIL!!!
ä€òžàŽrnnœnŒìŒœœœrbbnn‚ìbì‚rrn’lŒl`€’nžàŽrnnœnŒìŒœœœŽŽràbbnn‚bì‚rrnŒl`rnòžàŽrnnœnŒìŒœœœŽp€bbnn‚b|rrnŽŽlb‚‚nâþ‚rrnnœnŒìŒœœœŒŒbbnn€bì‚rrn‚ìŽŽlnîþ‚rrnnœnŒìŒœœœlŽpàbbnnŒb|rrn’lŒl`à’n|ržàŒ‚r’p`’‚’‚nâŒ’`brbì÷

Hardware initialized. Waiting for network.
mode : sta(18:fe:34:xx:xx:xx)
add if0
scandone
add 0
aid 6
cnt 

connected with WIFI_STATION, channel 1
dhcp client start...
ip:192.168.0.22,mask:255.255.255.0,gw:192.168.0.1
Fatal exception (0): 
epc1=0x402a0220
epc2=0x00000000
epc3=0x402a7183
epcvaddr=0x00000000
depc=0x00000000
rtn_add=0x401009e8

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16 
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0 
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0 
tail 12
chksum 0xde
csum 0xde
Œ‚ÿ
MEM CHECK FAIL!!!
ä€òžàŽrnnœnŒìŒœœœrbbnn‚ìbì‚rrn’lŒl`€’nžàŽrnnœnŒìŒœœœŽŽràbbnn‚bì‚rrnŒl`rnòžàŽrnnœnŒìŒœœœŽp€bbnn‚b|rrnŽŽlb‚‚nâþ‚rrnnœnŒìŒœœœŒŒbbnn€bì‚rrn‚ìŽŽlnîþ‚rrnnœnŒìŒœœœlŽpàbbnnŒb|rrn’lŒl`à’n|ržàŒ‚r’p`’‚’‚nâŒ’`brbì÷

Hardware initialized. Waiting for network.
mode : sta(18:fe:34:xx:xx:xx)
add if0
Fatal exception (0): 
epc1=0x402aee40
epc2=0x00000000
epc3=0x40103ed7
epcvaddr=0x402aef39
depc=0x00000000
rtn_add=0x402aef39

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32648, room 16 
tail 8
chksum 0x03
load 0x3ffe8000, len 1656, room 0 
tail 8
chksum 0x00
load 0x3ffe8680, len 6508, room 0 
tail 12
chksum 0xde
csum 0xde
Œ‚ÿ
MEM CHECK FAIL!!!
ä€òžàŽrnnœnŒìŒœœœrbbnn‚ìbì‚rrn’lŒl`€’nžàŽrnnœnŒìŒœœœŽŽràbbnn‚bì‚rrnŒl`rnòžàŽrnnœnŒìŒœœœŽp€bbnn‚b|rrnŽŽlb‚‚nâþ‚rrnnœnŒìŒœœœŒŒbbnn€bì‚rrn‚ìŽŽlnîþ‚rrnnœnŒìŒœœœlŽpàbbnnŒb|rrn’lŒl`à’n|ržàŒ‚r’p`’‚’‚nâŒ’`brbì÷

Hardware initialized. Waiting for network.
mode : sta(18:fe:34:xx:xx:xx)
add if0
scandone
add 0
aid 6
cnt 

connected with WIFI_STATION, channel 1
dhcp client start...
ip:192.168.0.22,mask:255.255.255.0,gw:192.168.0.1
Connecting to server 215.12.242.70...
Reading into SPI RAM FIFO...
Buffer fill 15999, DMA underrun ct 0, buff underrun ct 0
Buffer fill 31939, DMA underrun ct 0, buff underrun ct 0
Buffer fill 47975, DMA underrun ct 0, buff underrun ct 0
Buffer fill 63947, DMA underrun ct 0, buff underrun ct 0
MAD: Decoder start.
dec err 0x0101 (lost synchronization)
dec err 0x0103 (forbidden bitrate value)
dec err 0x0235 (bad main_data_begin pointer)
dec err 0x0235 (bad main_data_begin pointer)
Rate 44100
Buffer fill 70072, DMA underrun ct 2, buff underrun ct 0
Buffer fill 79838, DMA underrun ct 2, buff underrun ct 0
Buffer fill 91567, DMA underrun ct 2, buff underrun ct 0
Buffer fill 101333, DMA underrun ct 2, buff underrun ct 0
Buffer fill 113190, DMA underrun ct 2, buff underrun ct 0
Buffer fill 122956, DMA underrun ct 2, buff underrun ct 0
Buffer fill 131055, DMA underrun ct 2, buff underrun ct 0
Buffer fill 131041, DMA underrun ct 2, buff underrun ct 0
Buffer fill 131066, DMA underrun ct 2, buff underrun ct 0
Buffer fill 129313, DMA underrun ct 2, buff underrun ct 0
Buffer fill 124701, DMA underrun ct 2, buff underrun ct 0
Buffer fill 124279, DMA underrun ct 2, buff underrun ct 0
Buffer fill 121756, DMA underrun ct 2, buff underrun ct 0
Buffer fill 117145, DMA underrun ct 2, buff underrun ct 0
...
```

Spritetm, 2015-12-15:
Yeah, that's 99.9% sure a power issue then. Make sure you have a good 3.3V power supply that can source at least 500mA, and use good thick wires from it to the module. I'm closing this issue because the MP3 decoder seems to run, so this is not a software issue.

closed 2015-12-15 by Spritetm

milkpirate, 2015-12-15:
Thank you very much! And sorry that it was that profane in the end. But at least we were talking about that SPI RAM.

Btw: Should I be concerned about the decoder errors?
```
...
dec err 0x0101 (lost synchronization)
dec err 0x0103 (forbidden bitrate value)
dec err 0x0235 (bad main_data_begin pointer)
dec err 0x0235 (bad main_data_begin pointer)
...
```

Spritetm, 2015-12-17:
No, especially in the beginning these are perfectly normal. They can also be caused by ID3 tags, which (iirc) look like bad data to the decoder.