espressif/ESP8266_MP3_DECODER #13: SDK Versions and "section '.text' will not fit in region 'iram1_0_seg'"

SirCAS, 2016-02-21:
Hi,

What an awesome project! Keep it up :smile: 

I currently facing some issues when I try to compile the project. The compiler returns the following error:

`xtensa-lx106-elf-gcc  -L../lib -nostdlib -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -Wl,--Map=mapfile -u call_user_start -Wl,-static -Wl,--start-group -fdata-sections -ffast-math -fmerge-all-constants -fomit-frame-pointer -funroll-loops -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -lfreertos -llwip -ludhcp user/.output/eagle/debug/lib/libuser.a mad/.output/eagle/debug/lib/libmad.a driver/.output/eagle/debug/lib/libdriver.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 
/home/sircas/Source/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .output/eagle/debug/image/eagle.app.v6.out section '.text' will not fit in region 'iram1_0_seg'
collect2: error: ld returned 1 exit status
make: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1`

I'm using the _pfalcon/esp-open-sdk_ toolchain with Ubuntu 14.04 LTS.

The error indicate that there is a problem with fitting the code into the ESP module during the linking. So I checked the size parameter of the SPI FLASH RAM and it was correctly set at 1024. In an attempt to resolve the issue I've change this value to 4096.

`make COMPILE=gcc BOOT=none APP=0 SPI_SPEED=40 SPI_MODE=QIO SPI_SIZE=4096`

... But it didn't resolve the issue.

Then I started to wonder if the size of the SDK have increased considerable. Resulting in the binary blobs being to large that even 4M isn't enough.
I then changed the SDK to earlier versions.
Namely the following versions:
esp_iot_sdk_v1.0.1_b1 (based on the release date of the SDK and the initial commit of the MP3 project this could be the one used when developing the project)
esp_iot_sdk_v0.9.6_b1

Unfortunately the result was the same using the above SDK versions.

- Which versions of the SDK is known to work with this project?
- Does anyone have any suggestions for what I should try next?

Spritetm, 2016-03-01:
Sorry, I can't reproduce this. I am using the latest SDK (1.5.2) and esp-open-sdk and I have no problems compiling the project.

SirCAS, 2016-03-01:
@Spritetm Thanks for reaching out :hand:
I can't believe I've tried to play around with the playerconfig.h file sooner.

I did indeed modify the settings in the playerconfig.h file and this seems to be what cause the larger RAM usage. Which makes sense. This is my file for reference:

```c
#ifndef _PLAYER_CONFIG_H_
#define _PLAYER_CONFIG_H_

#define AP_NAME "sircas-test"
#define AP_PASS "sircas-test"
#define PLAY_SERVER "192.168.1.1"
#define PLAY_PATH "/sound.mp3"
#define PLAY_PORT 80
#define ADD_DEL_BUFFPERSAMP (1000)
#define ADD_DEL_BUFFPERSAMP_NOSPIRAM (1500)
#define DELTA_SIGMA_HACK

#endif
```

I've narrowed it down to the DELTA_SIGMA_HACK define. When it's enabled I get the mentioned error.
Using the PWM_HACK seems to work along with I2S. For now I'll just stick with the PWM hack, until I get some I2S circuit set up.

pali24, 2016-03-06:
Hi Spritetm, 
I have a similar problem, I can not run neither the make.sh file nor the make.... line.
Please help me, I am not a big programmer, but I like this brilliant project very much. :-)

Spritetm, 2016-03-07:
Seems the entire project is somewhat low on iram, I pushed a function to flash and that seems to have freed enough of it to make it possible to compile the delta-sigma code again.

pali24: Please open another issue, and please include some more info than 'I can not run make.sh nor make' if you want me to actually look into your problem.

closed 2016-03-07 by Spritetm