espressif/ESP8266_MP3_DECODER #15: Is it possible to read data from SD card instead of streaming server?

msurguy, 2016-03-25:
Thank you for this great project! 

What would it involve to read the sound from SD card instead of a streaming server?

Spritetm, 2016-03-26:
It would be possible, and would involve connecting an SD-card to the ESP8266, writing a driver to interact with it, and adding a filesystem driver and reader code to get the files from the SD-card. Hope that helps.

closed 2016-03-26 by Spritetm

nissimzur, 2016-03-26:
Since when GPIO15=0 it suppose to boot from SD card. I presume it has the SD card drivers already in its ROM firmware.
See http://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations
You can call my Skype nissim.test

Spritetm, 2016-03-26:
Sorry, we have not released any SD drivers. It is indeed probably possible to drive a SD-card over SPI. Plans are to release SD-card drivers for our next chip, the ESP32.