espressif/ESP8266_MP3_DECODER #16: Hardware PWM vs I2S

jobidon, 2016-03-28:
Great project, thanks for sharing!
The code lays out 2 alternative methods of generating the audio if the I2S IC is not available, but would it be possible to simplify this further using the built in PWM feature of the ESP8266 instead? Is the hardware PWM not fast enough to generate audio, unlike the PWM found on the arduino?

Spritetm, 2016-03-28:
Theoretically, yes. The hardware PWM generator is not DMA-fed, however, so using it would involve writing a timer interrupt firing at the sample rate to feed a byte into the PWM generator every time. This is probably possible, but it was out of the scope of this MP3 decoding example to actually implement it: I added the I2S PWM and delta-sigma examples because it only requires a slight modification to the code.

closed 2016-03-28 by Spritetm

Spritetm, 2016-03-28:
Oh, also: the ESP8266 doesn't have real hardware PWM, so it would actually involve clocking a timer at many times the sample rate. I was confused with the hardware delta-sigma-converter, which could be used in the fashion I described.

jobidon, 2016-03-28:
indeed, the PWM approach turns out to be pretty taxing on the processor.
The delta-sigma approach is very interesting. Do you know if/where 
there's documentation available specifically on how to use it with the 
ESP8266?

Spritetm, 2016-03-28:
No, sorry. I think it's mentioned somewhere in the ESP documentation, but I can't remember where.