espressif/ESP8266_MP3_DECODER #17: esp8266 newbie - where's the DMA precisely?

papadeltasierra, 2016-05-11:
I'm used to seeing DMA code that runs something like the following - can you point out where the equivalent code is in this project please?  Don't worry about stating the obvious; much better that I say 'I know that' to myself instead of 'but how do they get from X to Z, where's Y?'.

1. Set up a timer that will drive the DMA process
2. Set some memory up with the data to be thrown out
3. 'Attach' the memory to the output in some manner e.g. set registers to say 'PWM receives from DMA channel X'
4. Associate the timer with the DMA channel
5. Hit the 'go' button.
6. Either poll for status or wait for an interuppt that pops when the DMA has exhausted the data, assuming that it's not set to loop
7. Now stop the DMa if required and unroll the associations etc above.

FYI, what I'm hoping to do is to use this code as a basis to DMA stream data serially out of a GPIO port to a 433MHz transmitter to create a nice stable signal to send to a 433MHz receiver.  Doing this using a Non-OS CPU based routine doesn't work because the timings are way too variable :-(.

Thanks,
Papadeltasierra.

MswPaulDSmith, 2016-05-12:
Aha - seems all the magic is in mp3/driver/i2s_freertos.c.  All rather 'black box' though - is there any documentation for these incantations and interesting register writes?  For example, I think I might need to  either use a different set of calls or a different (slower) timer if I want to (mis)use the I2S interface for my  project.

Spritetm, 2016-05-24:
Sorry, there's not really any other information on the i2s hardware at the moment.

closed 2016-05-24 by Spritetm

papadeltasierra, 2016-05-24:
Jeroen,

Thanks for the feedback.  FYI the driver has a slight error – you can only divide the clock rate down by at most 63 x 63 not 127 x 127.  I’ve managed to (mis)use the interface by sending 32-bits for every true data bit meaning I’m running the clock at 32-times the speed I really wanted.  But it works a treat!

However next time I’ll probably use the HSPI interface instead as it looks like that can be done in a similar manner but with a lower clock speed.

Paul DS

P.S. I will release my code once I’ve had time to finish tidying it up.  Teenage children and their exams are slowing me down .

From: Jeroen Domburg 
Sent: Tuesday, May 24, 2016 7:33 AM
To: espressif/ESP8266_MP3_DECODER 
Cc: papadeltasierra ; Author 
Subject: Re: [espressif/ESP8266_MP3_DECODER] esp8266 newbie - where's the DMA precisely? (#17)

Sorry, there's not really any other information on the i2s hardware at the moment.

papadeltasierra, 2017-01-23:
For anyone working on this, I discovered yesterday that if you use the I2S/SPI interface in a 'send limited memory block, interrupt on completion' mode, the function of the GPIO pin seems to be 'reset to default' when the interrupt trips.  So when I wrote new data and triggered the send a second time, nothing seemed to happen but actually the DMA/I2S/SPI stuff was fine, but I needed to reprogram the GPIO pin again before starting.