espressif/ESP8266_MP3_DECODER #18: Connecting the SRAM

MrBuddyCasino, 2016-05-21:
I have an ESP8266 that uses one of those shielded silver AI-Thinker modules, where the flash chip isn't visible. Since no module currently sold seems to breakout the SPI pins used for the flash bus, only the HSPI ones, I'm clueless on how to connect the sram.

Can I just use the HSPI pins instead? If so, where and how do I change the pin definitions? I din't quite understand how spiram.c works.

nissimzur, 2016-05-21:
There is no SRAM. Only ram I the chip. 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

MrBuddyCasino, 2016-05-21:
That doesn't make any sense. I'm talking about the 23LC1024 chip thats required for buffering. I found a board that does have the SPI connectors (Olimex MOD-WIFI-ESP8266-DEV) and will try again.

nissimzur, 2016-05-21:
Please call my Skype nissim.test

The ESP8266 does not need SRAM to work. But you can hook and I2C or SPI chip to it 

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

nissimzur, 2016-05-21:
See https://github.com/OLIMEX/ESP8266/raw/master/HARDWARE/MOD-WIFI-ESP8266-DEV/MOD-WiFi-ESP8266-DEV_Rev_B.pdf

There is no 23LC1024
Only W25Q16BVSSIG flash. (NVM)

 

Kind regards,
Dr. Nissim Zur
Bluetooth Smart, IoT expert

 <http://www.elinistech.com/> CTO Elinistech 
Tel          0544 500635
Tel          03 5716699
Skype      <callto:nissim.test> nissim.test
Follow     <http://lnkd.in/TTnsGz> Linkedin  http://lnkd.in/TTnsGz
Email      nissim@elinistech.com

                <http://www.elinistech.com/> www.elinistech.com

Spritetm, 2016-05-21:
Nissimzur: Kindly refrain from trying to respond on a question you do not understand.

MrBuddyCasino: As far as I understand, pin 10, 13, 14 (the ones on the opposite side of the antenna) are connected to the SPI port. You should be able to connect the SRAM to those pins, plus GPIO0.

nissimzur, 2016-05-21:
Spritetm,
As I write above "The ESP8266 does not need SRAM to work. But you can hook any I2C or SPI chip to it . Also SRAM."
Read all reply befer flame responder. And I do design many ESP8266 projects.

Spritetm, 2016-05-21:
nissimzur: MrBuddyCasino asked a question that within the context of this project is perfectly clear; the readme (which isn't very long) extensively mentions connecting a SRAM chip to the ESP8266, and MrBuddyCasino has a very specific question with regards to that. You seem to not have read or understand this document, instead spouting generalities about the need or capability to connect any sram chip to the ESP8266, thereby demonstrating your ignorance of the question and the context around it. I ask you kindly to cease this behaviour as it adds noise to the issue list; when you have read the documentation/sources of this project and can show understanding of it, you're however more than welcome to come back and share your knowledge here.

MrBuddyCasino, 2016-05-21:
@Spritetm: thanks for the response! 
I found out that the ESP8266 has actually two SPI's. One is the HSPI, which is available on the GPIOs 12-15 (I can't spot pin 10 anywhere), and the other one is the SPI (not HSPI), used to interface with the flash ship. This is the one that is not commonly available on breakout pins. 

As far as I can see, to use the SPI one I need to either piggyback the sram on the flash, as you suggested, or use a dev board where those pins are available to the user.
With the Adafruit Huzzah, I think I can't do either of those things.

I just ordered the Olimex board, as the schematics suggest I can access the SPI interface (see https://github.com/OLIMEX/ESP8266/blob/master/HARDWARE/MOD-WIFI-ESP8266-DEV/MOD-WIFI-ESP8266-DEV_schematic.pdf).

The alternative would be to use the HSPI interface instead, but I suck at C and can't quite understand he code to make that change, if it is possible at all.

Spritetm, 2016-05-21:
The ESP12 on the Huzzah board actually does have the SPI connected, as I mentioned, but the Huzzah itself doesn't seem to route those pins any further. I think the Olimex board is a nice alternative. In theory it is possble to use the HSPI port instead of the SPI port, but implementing that takes some diving into the datasheets and other sources, so I think the Olimex board is your best bet indeed.

closed 2016-05-21 by Spritetm