espressif/ESP8266_MP3_DECODER #19: ES9023 MCLK

MrBuddyCasino, 2016-05-29:
First of all, thank you for this code, this is my first real MCU project and this gave me a major sense of achievement (I got it to work using the PWM output and the Olimex board).
It reminded me of the first detector radio that I built as a child.

I'm currently waiting for the ES9023 to arrive, and it seems I need an MCLK signal that is a multiple of the current sampling rate (e.g. 11,2896Mhz for 44,1KHz, there is a table on the ES9023 datasheet).

I guess I need some crystal/PLL to generate this myself? 
I couldn't find detailed information about this anywhere that a beginner like myself would understand. If you could say a few words or point me to a link, that would be most welcome.

milkpirate, 2016-05-30:
Hey,

it doesnt have to be exactly a multiple. The datasheet says:

> MCLK must be > 192*fs
(page 4)

Anyway. I had an old 50MHz crystla oscilator (see below, its not just a crystal it outputs a square wave - you can recognize them by them having four pins: GND, VCC, OUT, NC). It should suffice to connect OUT to MCLK (and GND/VCC for sure as well).

[image]

[image]

MrBuddyCasino, 2016-06-02:
Ah, hadn't noticed there was an asynchronous mode. Most I2S codecs I had researched don't have that, pretty handy. I'll wait for my crystals then and try my luck, thanks!

On a side note, I feel a little bad about opening issues here on Github about such questions. Is there a forum or something that you would prefer I use?

milkpirate, 2016-06-02:
I would suggest:

[http://electronics.stackexchange.com/](http://electronics.stackexchange.com/) (for any problem... stackexchange is a good idea)
[http://www.mikrocontroller.net/](http://www.mikrocontroller.net/) (german - pretty good though)
[http://www.esp8266.com/](http://www.esp8266.com/)

Spritetm, 2016-06-06:
milkpirate: Thanks for helping out here. For future reference: the board this code was tested on fed the ES9023 with a  40MHz crystal oscillator, but other values very probably will work as well.

closed 2016-06-06 by Spritetm

MrBuddyCasino, 2016-06-10:
@milkpirate: Thanks for the help!

I received my oscillators, attached the ES9023 and tested - it worked! I got sound via headphones, but very distorted, and stuttering even though there were no buffer underruns ("buff underrun ct 0"). The quality is much worse than with the PWM.

I haven't used any additional components besides a decoupling cap on the power supply, am I missing something?

These are the oscillators I used (50MHz and 25MHz), since they are breadboard-compatible: http://www.abracon.com/Oscillators/ach.pdf

MrBuddyCasino, 2016-06-10:
I think I get it now - the ES9023 has a VREG on pin 6, if there is no sensible resistor value between that and AGND it will hopelessly clip.

milkpirate, 2016-06-10:
@MrBuddyCasino: Hm... I'm facing similar problems - a lot of distortions. I used a 220k resistor so far. What value do you use? Are you running the - ES9023 at 3.3V?

MrBuddyCasino, 2016-06-10:
Nope, the ESP8266 doesn't get hot here. 
My Vreg experiments have yielded no results so far. It gets a little better when I pull DIF low (I2S left-justified) I think, but I'm not sure, maybe its just less loud.

MrBuddyCasino, 2016-06-11:
Pulling DIF low (10k Ω) helped, now it sounds like FM radio thats slightly out of range - a bit of crackling in the background.

I also added the capacitors from the data sheet application diagram, the 1uF cap on NEG is important. I also added the others, not sure how important those are though.

Some ES9023 schematics add resistors on the I2S lines, example:
http://1.bp.blogspot.com/-NDbWONqBv8Y/UMR7b5y2KDI/AAAAAAAAAkg/bYU5tVvYqyo/s1600/ES9023_SCHEMATIC_DAC_AUDIO.png

I tried that and it did little good, especially the 100Ω BCLK resistor was a bad idea, audio would sound very distorted. The whole thing is very sensitive to interferences it seems due to the high frequencies involved, having a background in electrical engineering would probably be useful...

sapir, 2016-12-12:
Regarding MCLK - I just tried setting ESP8266's GPIO0 function to CLK_OUT and using that as MCLK, and it seemed to work with no additional oscillator. Probably noisy and not compatible with using SPIRAM but works for me for now.