espressif/ESP8266_MP3_DECODER #20: ESP-12E gets extremely hot

milkpirate, 2016-06-10:
Hey,

like in [https://github.com/espressif/ESP8266_MP3_DECODER/issues/19](https://github.com/espressif/ESP8266_MP3_DECODER/issues/19 ) discussed my setup works - kind of. While running the system the output is pretty distorded and the ESP gets extremely hot (burned my finger touching it). I was also wondering if it has some thermal protection? Could the software do some wired stuff? If not I have an idea why this could happen: Please have a look at the pictures below.

[image]
[image]

I always thought these SMD inductiors are shielded but since the inductor of the buck converter is seemingly right below the ESP, I think it could induce some current that results in this heating. Is that possible?

MrBuddyCasino, 2016-06-23:
Spritetm is probably busy right now with the launch preparations of the ESP32. I may have a suggestion for the distortion issue though. 

After I suffered lots of crackling with the ES9023 too, I ordered two assembled I2S boards from China, to see if it is an issue with my hardware or the chip itself.

The one with the ES9023 had less issues than my setup, but still there was some crackling. The one with the PCM5102 did not have this problem, also it doesn't even need an MCLK signal (see http://www.ebay.de/itm/331591969853). Just make sure to undefine ALLOW_VARY_SAMPLE_BITS and set the jumper to "I2S", not "LJ" (=left justified) .

I feel the PCM5102 is a bit more picky than the ES9023, if the connections aren't 100% alright or not theres enough voltage it will stay silent, but there are no distortions anymore. The only problem remaining is that sometimes the pitch is not quite right, but thats a software issue and a result of the attempts to correct the clock drift I think.

MrBuddyCasino, 2016-06-25:
I just tried another board, namely the MAX98357A (breakout: https://www.adafruit.com/product/3006). Works as well, doesn't need an MCLK and no separate amp required, but the DIN line is quite sensitive - if it sounds distorted, thats probably where problem lies.

I think I'll build my project with this one.

MrBuddyCasino, 2016-06-28:
FYI, I managed to reduce the distortion a lot by adding 1nf caps between the data and bclk lines and ground. It seems I2S is not an easy protocol to wire, signal distortions (overshoot, noise, reflections etc.) can and do occur easily due to the high frequencies involved. I'll get myself a scope (Rigol DS1054Z) to fully diagnose the issue, and I try to learn about good analog design practices.

Heres more reading: 
http://www.diyaudio.com/forums/digital-line-level/236932-i2s-problems.html
www.x2y.com/filters/TechDay09kr_hpa_Track2_1_Precision_Analog_Designs_Demand_GoodPCBLayouts _JohnWu.pdf

Spritetm, 2016-08-26:
Closing this because the discussion is somewhat old and this is not a software issue.

closed 2016-08-26 by Spritetm