espressif/ESP8266_MP3_DECODER #21: xTaskCreate sets usStackDepth to e.g. 2100 for 2100 bytes (not 2100 words)

DanBennettDev, 2016-06-12:
Hi,
I've been reading the freeRTOS "practical guide" and it states that the parameter usStackDepth in xTaskCreate specifies the stack depth in words - but your comment in user_main, in the function tskreader, states that the value 2100 is used to request 2100 bytes. I understand that the ESP8266 is a 32bit architecture, so is this a typo (for 8400 bytes), or does the specific behaviour of the version of freeRTOS you use here differ from that documented in the practical guide? (or, third option - have I just completely misunderstood what's going on in some other way?!)

Thanks
Dan

MrBuddyCasino, 2016-06-28:
I checked the stack memory consumption, and the values used are reasonable - theres a little headroom left, but this is definitely not a typo. If you need more memory, I think you can get maybe 200-300 bytes at most by changing the stack sizes.

DanBennettDev, 2016-06-28:
I think we're perhaps at cross purposes. I don't think the amount of memory allocated is inadequate. I'm asking if the implementation of RTOS used is the same as that in the practical guide. 
I probably could have phrased my question more clearly:

The RTOS practical guide states that uStackDepth specifies the stack depth in *words* not *bytes*. It follows from this that for a 32 bit machine a  uStackDepth of 2100 will mean 8400 bytes. However, in the code here (in user_main)  the function tskreader uses a uStackDepth of 2100 and comments that by saying that it requests 2100 bytes (as opposed to 2100 words, or 8400 bytes) . Is the comment here correct in refering to 2100 *bytes* rather than *words* (in which case I suppose the freeRTOS version used specifies bytes, not words in uStackDepth), or is it a typo?

Sorry if this seems pedantic - it's just that I'm considering build on top of this code, and I saw in the readme that the version of freeRTOS here may be altered in some ways for efficiency, so want to be sure that I can continue to assume that uStackDepth is specifying words (as in freeRTOS docs), not bytes. As I say - it is also possible I am just misunderstanding (I'm very new to both freeRTOS, and embedded development in general) if so apologies!

Spritetm, 2016-06-28:
Sorry for not having replied yet. I don't have time to look into this right now, but I think I made a thinko when I wrote the 'bytes' there; it most probably should indeed be 'words'. The stack depth was figured out mostly by trial-and-error, so I'm pretty sure the values themselves are right, whatever the unit.

DanBennettDev, 2016-06-28:
No problem - I assumed busy with the ESP32, and it's a very minor issue. It works, so as you say the values themselves are unlikely wrong.

So thanks - that's enough reassurance for me that I can follow the RTOS practical guide in building on this. 
Thanks in general for this repo - v exciting project which raises loads of possibilities.

Spritetm, 2016-08-26:
Fixed the comment in ed4e4e4

closed 2016-08-26 by Spritetm