espressif/ESP8266_MP3_DECODER #23: Setup Help

bickyb, 2016-07-12:
Hi,
I am new to this project but I am very interested in being able to play MP3s off of my ESP8266.  I am actually using the Huzzah from Adafruit: https://www.adafruit.com/product/2821

I have few questions:
1) I am new to github so it might be I don't know how to navigate it well, but is there a step by step guide somewhere of how to get this project up and running?  If not, I would love to help document the procedure if someone can hand hold me.
2) Would this setup be allowed to play Mp3 files on demand from a web server?  I see that there are mentions of being able to play from an Icecast server.  I just want a setup that can play different MP3s from my ESP8266 on demand from start to finish (e.g., not midstream)  Would this project work for me and would I have to have an Icecast server involved?
3) Is there a fritzing diagram or something that shows how to wire up the ESP8266 to implement this project?

phlemoine, 2016-08-25:
I'm also using the adafruit esp8266 , but the breakout version (https://www.adafruit.com/products/2471)  not the feather version as mentioned above. I am struggling to get "sound", my best attempts so far yielded to noise only.
I still have couple of experimentation to run, however I start to run out of idea of why it's not working for me. 
The breakout board has pull up/down on gpio 2 and 15 and a level shifter in gpio 3. These are the 3 gpio required for the i2s port. 
I am ready to remove these pull up/down on gpio 2 and 15, and I can already bypass the level shifter of gpio 3 by soldering a wire directly on the pad.
Before doing so, I would like to have some feedback on how can these pull up/down impacts the LRC and BCLK lines?

Spritetm, 2016-08-26:
bickyb: All the documentation is in https://github.com/espressif/ESP8266_MP3_DECODER/blob/master/README.md

phlemoine: what codec are you using?

phlemoine, 2016-08-26:
I'm using this add on board :  MAX 98357A  at : https://www.adafruit.com/product/3006

Spritetm, 2016-08-26:
In mp3/user/playerconfig.h, have you tried commenting out the line #define ALLOW_VARY_SAMPLE_BITS ?

phlemoine, 2016-08-26:
Yes. I tried several combination including the mp3 streams and mp3 file, the VARY_SAMPLE_BITS on/off, I also tried remapping to different pins (not sure that iomuxing even allows that tho). I have ordered an ESP8266 module (without the additional components that usually comes with any breakout boards) so that I can verify my theory of the pull up/down on your 2 and 15

phlemoine, 2016-08-27:
update to my investigations. i have used the sample code at https://github.com/SuperHouse/esp-open-rtos/tree/master/examples/i2s_audio and can prove the HW is working fine (playing a pre-flashed wav file in SPIFFS). this helps tremendously with restricting the investigations to SW (or build or flash) issues.

phlemoine, 2016-08-27:
ok, i am not sure what was wrong (and that is frustrating :-) ) but i eventually got this project to work. thanks for taking time to respond. 

it works with ALLOW_VARY_SAMPLE_BITS commented out (and  the MAX98357A breakout)
also PWM_HACK, DELTA_SIGMA_HACK commented out (naturally)
and  FAKE_SPI_BUFF enabled since i dont use the flash chip
i am using icecast.omroep.nl and based in the US

i might have (inadvertently) combined options in such a way that created noise and suspected HW issues for too long

hope these notes will help others running into any issues

Spritetm, 2016-08-27:
Okay, glad to hear you have it working now.

closed 2016-08-27 by Spritetm