espressif/ESP8266_MP3_DECODER #32: Schematics

josephernest, 2017-10-01:
This project looks great, could you share a schematic for it?
Can I build one without any PCB, but just ESP8266, the SSOIC DAC and SSOIC RAM (and probably breakout PCB for SSOIC)? 
Thanks in advance if you can share a photo, or anything that would help the prototyping.

blueowl04, 2017-10-18:
There's a description in README. But it would be definitely helpful if a hardware scheme is created.
Anyway, I have found some references that might be useful:
https://hackaday.com/2015/06/06/esp8266-as-a-networked-mp3-decoder/
https://www.youtube.com/watch?v=3euFJSkDcFI
https://www.youtube.com/watch?v=JsRsNDIBQsM

gr01d, 2018-12-28:
Here is my schematic:

Video demonstration:
Choose your preferred site, either Youtube or Peertube they both have the same video.
https://tube.22decembre.eu/videos/watch/c426fb1e-d194-416d-8985-37dbf800e68d
https://youtu.be/fL9VkbOUq1s

josephernest, 2019-01-02:
Thanks @gr01d for this video, very interesting!
What would be the minimum number of parts to have better sound quality?

esp8266 + DAC (which one to connect on the ESP?)

gr01d, 2019-01-21:
I cannot answer this question, I do not have a DAC or SPI Flash RAM but only an ESP8266 (nodemcu v1) board.
The SPI Flash Ram is needed to avoid audio buffering (lack of internal RAM on esp8266) and the DAC to make the audio quality good (it also has the codec to decode the audio). IIRC